ImpulseRC/OSD #87: Altitude in statistic are relative to sea level

slackista, 2021-04-13:
I have BMP388 and GPS with V90.27 v2 board. In post flight statistic max an min altitude are probably relative to sea level.  Values are to big. In previous versions of firmware problem did not occur.

awolf78, 2021-04-17:
I have uploaded V91.27 that should fix this issue and #88. I cannot test it myself unfortunately, please let me know if this fixes it.

slackista, 2021-04-18:
Thank you for fast fix. Tested changes today. Statistic are fine now. New issues found:
* GPS ~7/10 power ups, no GPS via crsf telemetry and no beep from buzzer(distinctive gps beep). OSD was OK. 
* For the first time was very hard to get fix. After half hour, lots of unplugged battery finally GPS got fix. Probably because no setup init command to gps was send. 
* ~1/10 power ups no GPS on OSD.
Previous firmware also had issues with random lack of gps crsf telemetry feed. But initialization problem was more noticeable now.

slackista, 2021-04-18:
Thank you for fast fix. Tested today, and worked fine.
One more thing, in crsf telemetry feed altitude is relative to sea level ;)

awolf78, 2021-04-18:
OK, please flash V91.27 again (I replaced the binary with one that fixes this).

slackista, 2021-04-18:
Fixed. thanks!

closed 2021-04-18 by slackista