ImpulseRC/OSD #89: RTH not working

1ALeRT2, 2021-05-17:
Hello.  I have wolf v3 pdb + kiss v2 fc + matek m8Q.  Configured wolf v3 pdb - GPS coordinates, speed, altitude, distance to home are shown in OSD.  In addition, the pdb settings are as follows: 3d speed is off, the altitude reading from the home position, in the gps settings 10KHz, the magnetometer is calibrated and shows correctly in flight, a switch for rescue (RTH) is assigned to fc in the high position, to the same switch is assigned to align in  position middle.  Return to home does not work when RTH is turned on.  What could be the problem?

awolf78, 2021-05-17:
What firmware do you run on the KISS FC? Did you configure a GPS in the KISS GUI (should not be configured if connected to the Wolf)?

1ALeRT2, 2021-05-17:
Firmware 1.3p-rc46f. kiss v2 has no built-in osd

1ALeRT2, 2021-05-17:
does it mean that i don't need to configure the rth switch in the gui?

1ALeRT2, 2021-05-17:
what then needs to be configured in kiss gui?

1ALeRT2, 2021-05-20:
I understand there is no solution to the problem with gps?

awolf78, 2021-05-21:
Yes there is, but it requires a firmware update. Cannot make any promises when I can fix it, but I will.

1ALeRT2, 2021-05-21:
I will wait

awolf78, 2021-05-23:
Live now