ImpulseRC/OSD #91: Last firmware / GPS

ThomSloop, 2021-07-03:
Hello, it’s very strange : When i’m running the 86.12 firmware, the GPS works fine, but when i’m updating to the 90.xx or earlier, GPS doesn’t work again.

Any idea ?

Thanks for your help.

awolf78, 2021-07-03:
Check if the 25Hz mode is activated - that only works on M9N GPS modules.

ThomSloop, 2021-07-03:
I have a BN-220, so it’s a m8q module.

awolf78, 2021-07-03:
That one doesn't support 25Hz either

ThomSloop, 2021-07-03:
So, I stay on the 86.xx version or I change my gps to move to the earlier version ?

awolf78, 2021-07-03:
You can upgrade to the latest version just make sure the 25Hz mode is turned off.

ThomSloop, 2021-07-03:
Thanks but already tested with no result. So i prefer staying with the 86.xx version who work perfectly… Sometimes I’m trying new releases, but no one work at this time.

awolf78, 2021-07-03:
When you change to 10Hz you need to make sure you save and then power cycle.

ThomSloop, 2021-07-05:
Ok; I’ve tried this and I have the coordinate into the fpv mask but when I discover new sensors on the tango 2, she see everything except the gps.

awolf78, 2021-07-05:
What version are you running ImpulseOSD/KISS FC?

ThomSloop, 2021-07-06:
The last one for both; so I think it’s 92.xx for impulse and 46.xx for kiss