ImpulseRC/OSD #93: OSD completely disappears

udyux, 2021-08-22:
I have 3 nearly identical Apex. All have the FETtec F7 and 45A (except one with the Flyduino ESC) running 4S, regulator board, Apex OSD, Unify 5V V3, Falkor V2. Both the FETtec ESC builds have spike absorber and cap. All have same/latest FW and identical configs.

All are wired up the same with an additional ground running from the FC to the ground pad on the mic side of the OSD. All wires run above the FC.

On one build (45A ESC) the OSD completely disappears on throttle punches or sharp moves. At first it would only happen on 1/5 packs but lately it's on every pack. Power cycle brings it back (but it's lost consumption from before the power cycle, doesn't ask if I want to continue the last battery). I also noticed the "heartbeat" LED on the OSD isn't blinking anymore when the OSD is gone but the other lights are still on.

It's never happened on the two other builds. Any ideas?

awolf78, 2021-08-22:
How is the FC connected to the ESC? With the wire that came with the ESC? In that case remove the VBAT between FC and ESC. You are already getting VBAT from the regulator board to the FC.

udyux, 2021-08-22:
I've got the FETtec 45A feeding the FC and pulled the VBat and Gnd from the regulator harness to the FC. I've done this on both of the builds using that ESC. Could this be the cause? (Haven't experienced the issue on the other)

awolf78, 2021-08-23:
You need the GND from the ESC to FC - but remove the VBat between ESC FC.