ImpulseRC/OSD #94: ODS board overheating then video doesn't work

Elodin12, 2021-09-14:
I recently had a minor crash and the video cut out on my apex. after the crash, I could see the video and the osd boot logo, and then in about 10 seconds the video would kinda waver and said out to black. when I checked the connections all the wires had a solid connection but the big chip next to the microphone (I think it's the voltage regulator or something)would heat up to untouchable temperatures in a matter of seconds. The osd also said it did not have a connection to the FC on startup. I tried the rescue flash after resoldering all the connections just to be sure. Now I don't even get those few seconds of video I had just after the crash. Any idea how I could go about fixing this or should I just contact who I bought the Apex from to see if I can get a replacement? Thanks

Elodin12, 2021-09-14:
Is it possible to connect the yellow cam wire to the vtx directly to make sure that I diagnosed the osd as the problem Apex osd board?

sblakemore, 2021-09-15:
Hi, if the voltage regulator is overheating and shutting down this is usually caused by a short somewhere which is drawing too much current for the regulator to handle.

This github is for software issues though, for hardware support please contact ImpulseRC directly.

closed 2021-09-15 by sblakemore