ImpulseRC/OSD #98: GPS SBAS Setting

slackista, 2022-04-10:
There is a bool config option for SBAS.
1) What exactly is doing ON option? Sending ubx sbas auto?
2) I assume "Off" option sending "sbas disable" command. Is there option to **not touch** SBAS configuration in gps unit?

sblakemore, 2022-04-27:
1. Yes, it sends sbas auto.
2. Your assumption is correct, and I'm afraid there is no option to not touch the GPS config.

closed 2022-04-27 by sblakemore

slackista, 2022-07-18:
Reopening. Finally find cause of serious gps bug. After a while SV in use is dropping to 5 from 16(max @10hz). This RTH is impossible. Using latest Neo-M9N unit.  

**Description**

Apparently your SBAS config that is sending to the ublox unit contains configuration "CFG-SBAS-USE_INTEGRITY": "TRUE"
The UBLOX manual says:
>When SBAS integrity data are applied, the navigation engine stops using all signals for
which no integrity data are available (including all non-GPS signals). It is not recommended
to enable SBAS integrity on borders of SBAS service regions in order not to inadvertently
restrict the number of available signals.

In Europe there is EGNOS. EGNOS augment **only** GPS(https://defence-industry-space.ec.europa.eu/eu-space-policy/egnos_en) thus unit is dropping all other constellations i.e. GLONASS, Galileo, BeiDou and unit reports only 5 SV.  Which is unacceptable for RTH in Kiss, minimum is 6 SVs.

**Possible Solutions**
1. Create option to not touch SBAS configuration.
2. Change CFG that set CFG-SBAS-USE_INTEGRITY": "FALSE" or do not touch that option, most units has that in "false"