ImpulseRC/OSD #100: Black icon near

SGr33n, 2022-04-29:
Hi!
On my Apex OSD, I have black squares instead of timer icon and crsf signal.
How can I solve this?
Thanks!

[image]

krunked-fpv, 2022-04-29:
Update font try again.

SGr33n, 2022-04-29:
Already did this, nothing changed :(

awolf78, 2022-04-30:
Flash with "full chip erase"

SGr33n, 2022-04-30:
That's just what might have broken it, I did it yesterday before I had this issue. Can I safely proceed? I read if I do too many times it's possible that the server can blacklist my S.N. and don't activate it anymore automatically.

awolf78, 2022-04-30:
You are good for a about 5 times for the full chip erase - if you get blacklisted ImpulseRC support can remove the blacklist. I would try one more time flashing with a full chip erase under the following conditions:
- Use a fully charged Lipo connected to the board
- Disconnect as much as possible from the board

If that doesn't help I would contact your dealer and discuss the issue with them.

SGr33n, 2022-04-30:
Thanks, I will try!