ImpulseRC/OSD #101: ELRS and Apex OSD

Linuxbarista, 2022-05-30:
Has anyone tried getting the RSSI or LQ from ELRS to work in the OSD? I've tried doing the "old method" of putting a telemetry sensor (1RSS) data on an input channel of the radio and pushing that to a mixer channel. I can't seem to get it to read out correctly. 

Hoping there is a guru here that can explain what min/max PPU I should set and what I'm doing wrong on the radio input mixer side? Basically followed Bardwell's method from a couple of years ago. https://youtu.be/t-evOAS9Mkg I do know that ELRS runs from -128db - 0db. I'm not too familiar with how FRSky RSSI works. I believe it's 0db - 100db. I did try playing with different "scale" and "weights" on the inputs. Also tried playing with the "weight" and "offset" in the mixer section. I got it working somewhat but it wasn't very accurate. I know ELRS is outputting RSSI or DB on AUX 12 and I believe LQ is on  AUX 11. 

I'm guessing this is prob not going to be possible until channel mapping shows up in ELRS like Crossfire and Tracer where you can assign a channel to the LQ etc...

I know all this is super easy in betaflight OSD and Kiss OSD. I really like using the solid regulators built into this system and the ease of camera and VTX wiring positioning. Currently on Tracer at the moment but have ELRS gear I'm looking to swap in. 

Thanks for anyone's help in advance!

Linuxbarista, 2022-11-14:
Anybody?

slackista, 2023-03-15:
This is dead project apparently.